WebReflection/benja #1: License for Use.

seedceo, 2017-01-24:
Thanks for your nice work to create this awesome OS.

I'm looking for the license but found nothing, could you please provide some information about this?

Thanks you so much.

WebReflection, 2017-01-24:
You are right, I haven't specified any license because I am not sure which one would be more appropriate.

[ArchLinux ARM](https://en.wikipedia.org/wiki/Arch_Linux_ARM) apparently has GPLv2

[Electron](https://github.com/electron/electron) has MIT style license

However, software included in ArchLinux has various licenses.

I think this cannot have a license more permissive than its two major components it ships with, so while my code usually is under MIT Style License, there is a GPLv2 here which is redistributed.

Do you have any suggestion?

seedceo, 2017-02-06:
Thanks for your answer.

closed 2017-02-06 by seedceo