WebReflection/benja #3: Chip

CrackerakiUA, 2017-02-03:
Have you tryed https://getchip.com/pages/chip
Looks promising. Do you have any comments on it?
probably this OS will work on it.

WebReflection, 2017-02-03:
I have one, and realised only today they finally managed to bring ArchLinuxARM to chip.

http://www.chip-community.org/index.php/Archlinuxarm_on_chip

I'll give it a try, I love that device!

WebReflection, 2017-02-06:
I'm afraid the ArchLinuxARM install does not support the HDMI output and I couldn't test properly in there.

However, I've managed to start the app on regular Debian after 1 hour to update everything and install everything else.

As result, this board doesn't seem to be a good target for various reasons: the GPU through drm is not recognized so it's entirely software rendered and pages are basically unusable, and within the current Desktop env there's not enough RAM so I have problems even starting through `npm start` instead of just `electron index.js` installed as both global and local due some issue in finding files.

I think for now I'll flag this as won't fix because the time it'll take to have whole Benja OS up and running won't bring expected results (if not even official Debian with latest dedicated dtb can use HW acceleration I strongly doubt my install would do).

The curse of closed source GPU drivers 😞

closed 2017-02-06 by WebReflection