WebReflection/benja #4: Keyboard and/or mouse problem...

Snille, 2017-02-03:
Any news in the issue to be able to start the OS without a input-device connected?
I'm thinking of using this great OS for photo frames on RPi3:s (mostly). :)

Thank you for bringing this to life!

WebReflection, 2017-02-03:
Have you tried to SSH to the PI and `sudo pacman -Syu` or, if available, `yaourt -Syua` ?

If that hasn't solved the problem, all I can say is that I'm testing RPi2 image without using the broadcom blob since Mesa apparently supports already pretty well VC4 GPU.

I'm having crashes on RPi3 and ARM64 version, which is not officially supported that, so please bear with me I'll update RPi2 image with latest "everything" and quite possibly without any issue like that.

Snille, 2017-02-03:
Great to hear! :) I tried the first time to with the `sudo pacman -Syu` but still it "hang". However, SSH worked as it should. But then I broke it anyway by trying lot´s of stuff. :) So I've reinstalled and are just running the update now... I'll play around a bit more. I'm for the moment running it on a RPi 3 Model B without crashes at least. :)
Looking forward to the updated version. It's lot's to update from a new install. :)

Thank you.

WebReflection, 2017-02-03:
Current image uses a specially built Weston for the Raspberry Pi to bring in HW acceleration.

Latest ArchLinuxARM 4.9.x has most RPi2 HW covered already and the weston-pi has been removed in favor of weston on mesa.

I don't want you to do too much testing but one easy thing would be to `sudo pacman -Rsc weston-pi` and then `sudo pacman -S weston` chosing Mesa, if asked, as main libgl or video library, and see if that solves anything.

Snille, 2017-02-03:
Ah, ok. :) I'll get on it!
I got this when upgrading (I think I missed that on the last update):
ttf-dejavu: /etc/fonts/conf.d/20-unhint-small-dejavu-sans-mono.conf exists in filesystem
ttf-dejavu: /etc/fonts/conf.d/20-unhint-small-dejavu-sans.conf exists in filesystem
ttf-dejavu: /etc/fonts/conf.d/20-unhint-small-dejavu-serif.conf exists in filesystem
ttf-dejavu: /etc/fonts/conf.d/57-dejavu-sans-mono.conf exists in filesystem
ttf-dejavu: /etc/fonts/conf.d/57-dejavu-sans.conf exists in filesystem
ttf-dejavu: /etc/fonts/conf.d/57-dejavu-serif.conf exists in filesystem

Had to (re)move the files (not owned by any package). Don't know if those are yours?

I'll try out the driver soon. :)

WebReflection, 2017-02-03:
https://www.archlinux.org/news/ttf-dejavu-237-will-require-forced-upgrade/

You need `pacman -S --force ttf-dejavu` first, then `pacman -Syu`

Snille, 2017-02-03:
Hehe, thank you.
Not been playing with arch a lot. :)

WebReflection, 2017-02-03:
these things happen rarely but when it happens it breaks everything ... I guess it's the only side effect of being on a rolling based distro. We've got all latest and greatest software for the Pi on it though!
🎉

Snille, 2017-02-03:
Hehe, yes. :) Ok, After the update it electron "crashes". I also tried 'sudo pacman -Rsc weston-pi' (target nor found)and 'sudo pacman -S weston' is already installed. It reinstall it. But still no go. :)

When trying to update electron only I get:
```
[benja@benja ~]$ sudo npm -g update
npm ERR! addLocal Could not install /build/semver/src/semver-5.3.0.tgz
npm ERR! addLocal Could not install /build/npm/src/npm
[benja@benja ~]$
```

So, there is some problems indeed. :)

WebReflection, 2017-02-03:
I won't probably be able to fix this for the next 2 weeks. Conf to prepare, full time job, 1 week end snowboarding, sorry.

It's going to take a little while unless you manage to fix it though ArchLinux. After all there's no extra magic involved here, Benja OS is a configuration, not a distro a part. ;-)

Snille, 2017-02-03:
No problem! Don't worry about it, no hurry. I'll fiddle around with it. It's always fun to learn new stuff.
you just do it when you have the time. :)

Thank you.

WebReflection, 2017-02-03:
So ... I've made some change, and switched to Xorg and native Mesa, since weston, or weston-rpi, were the one unable to start without an input.

The new iso for the RPi2/3 just lande on my S3 instance, click download and let me know how that worked.

https://benja.io/

Snille, 2017-02-03:
I get access denied when trying do dl (this: https://s3.amazonaws.com/benja.io/iso/benja-rpi2.zip :).

```
This XML file does not appear to have any style information associated with it. The document tree is shown below.

<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>70580A663A3D0623</RequestId>
<HostId>
QadbA0t2PYiseobS7Lo5x+/wZZcpo+MAQx3hUAmJh7l/XVLb34/wct178cCJlQj013JjpxxOBYU=
</HostId>
</Error>
```

WebReflection, 2017-02-04:
please try again

Snille, 2017-02-04:
Yepp, working! Thank you!
Oh! You added lots of updates as well? :) Now it's only 5 needed when updating.
Question, when switching to X again, does that mean we loose the Hardware accelerated display stuff?
Great work!

WebReflection, 2017-02-04:
No, it means ther's no XWayland intermediate layer.

Weston-RPi was a Weston build around closed source Broadcom VC4 GPU drivers.

Meanwhile, Mesa 13.X started supporting the RPi out of the box so there's no need to use the old blob, hence Weston.

Ideally, as soon as Electron will work on Wayland, I'd be more than happy to drop Xorg and use Weston only because it's lighter and faster. Xorg, on the other hand, has better driver supports being very old platform.

Tough to say today what it will be ... but hey, even WebGL works like a charm now!

Check this out:
https://twitter.com/WebReflection/status/827875726359687168

closed 2017-02-04 by WebReflection

Snille, 2017-02-04:
Looking GREAT! :) Love it!