WebReflection/benja #5: Video playback

GegeFR, 2017-03-05:
Hi, I might try this on pi 3 (need to do a kiosk ui).
Is the video playback HW accelerated ? I'll have to read 1080p@30 or 720p@30.

WebReflection, 2017-03-05:
haven't tried but I don't see why it wouldn't be. let me know how that goes.

GegeFR, 2017-03-05:
I tried and sadly it is not HW accelerated. Tried to read the 1080p Big bunny Buck video from local node instance and it lags a lot (sd card is fast enough since it works when I'm doing it locally with npm start).

And a pi2, with it's native player (omxplayer) can easily play it.

I'm looking for a way to do a nice HW accelerated kiosk UI, able to play 1080p videos. I was looking into android (on banana pis or pine 64) but i was bothered by the fact that it has issues handling non-16:9 resolutions, and that android won't  allow me to easily use gpios, i2c.

WebReflection, 2017-03-05:
uhm ... surely omxplayer would perform better than Chromium behind Electron, but I'm afraid the strength and the weakness of Benja is actually Electron itself: it is what it is.

Have you tried any Qt alternative? Or actually even GTK and its webkit, it's not too bad but I don't know if QtWebView would be better, for sure I know QtQuick has a native video player and it should perform as good as in anything else since it's the UI behind many media servers.

Give it a try, sorry benja wasn't probably for you!

closed 2017-03-05 by WebReflection