WebReflection/benja #6: Cache update

alexbenza92, 2017-03-13:
Hello, i have a problem when i modify index.html the changes aren't appearing on the screen even if i reboot the device. What should i do ? Thank you!

WebReflection, 2017-03-13:
have you tried cache invalidation?

try to load this url in your `index.js` file

```js
.loadURL('http://localhost:8080/?_=' + Math.random())
```

If that solves, it might be that `tiny-cdn` does its job too greedly, forcing cache a bit too much.

Please let me know, thanks.

alexbenza92, 2017-03-13:
Solved thank you!

Benza Alexandru

Web Developer

Tel: 0757 080 888

alex.benza@mail-dms.com

On March 13, 2017 at 11:54:21 AM, Andrea Giammarchi (notifications@github.com) wrote:

.loadURL('http://localhost:8080/?_=' + Math.random())

WebReflection, 2017-03-13:
that's something puzzling me, tidy-cdn is awesome for prod but painful in development.

Maybe I should disable cache completely through an env flag ... I'll think about it, glad I've sorted your problem.

Bear in mind same thing might be needed for external files you are loading through that page.

closed 2017-03-23 by WebReflection