WebReflection/benja #7: How to access the terminal ?

manju4ever, 2017-03-23:
I was wondering how do we access the arch linux terminal to install drivers and stuff ?

WebReflection, 2017-03-23:
ssh benja@192.-68.1.xxx

Password: benja

The ip to ssh into is the one shown first time you boot your device connected to any dhcp

closed 2017-03-23 by WebReflection

maertz, 2017-06-29:
Is there also a root acc.?

WebReflection, 2017-06-29:
root/root I guess

maertz, 2017-06-30:
Indeed as you mentioned in your docs but at least within the RPI ISO it doesn't work.

WebReflection, 2017-06-30:
root acces via SSH might be disabled .... but you have benja user, with sudo privileges ... so, what are you trying to do that's not already possible?

maertz, 2017-06-30:
I tried to rebuild your sample and use the [GPIO pins](https://www.youtube.com/watch?v=AsVvoy2K8d8) but mine says johnny-five and raspi-io aren't available so I tried to install them the hard way.

WebReflection, 2017-06-30:
the hard way? what `sudo npm install -g johnny-five` didn't work?

maertz, 2017-06-30:
Yea I'm basically trying every possibility I can think of (the hard way) ^^ just tried your command on a fresh installed ISO - didn't work, here is the output (RPI Model 3 rev 2).
```
Last login: Fri Jun 30 12:05:15 on ttys000
You have new mail.
maertz $ ssh benja@192.168.0.66
benja@192.168.0.66's password: 
 
  Raspberry Pi 2 
 ┌────────────────┐
 │ ╔╗ ╔═╗╔╗╔ ╦╔═╗ │ Bootable
 │ ╠╩╗║╣ ║║║ ║╠═╣ │ Electron
 │ ╚═╝╚═╝╝╚╝╚╝╩ ╩ │ Node JS App
 └────────────────┘
  https://benja.io  0.3.0
 
[benja@benja ~]$ sudo npm install -g johnny-five

> serialport@4.0.7 install /usr/lib/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v51-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@7.5.0 (node-v51 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.5.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/johnny-five/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/lib/node_modules/johnny-five/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/7.5.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/johnny-five/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:192:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.45-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/lib/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v7.5.0
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:192:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:890:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.45-1-ARCH
node-pre-gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v7.5.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release' (1)
/usr/lib
└── johnny-five@0.11.3 

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: serialport@4.0.7 (node_modules/johnny-five/node_modules/serialport):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
[benja@benja ~]$ 
```

WebReflection, 2017-06-30:
OK, first of all, IIRC johnny five should be already there. I guess that's not actually the case.

Before you can build stuff on ArchLinux you need dev tools.

However, I suggest you update the whole system first.
The current ISO is quite old, you probably have to update [this way](https://www.archlinux.org/news/ca-certificates-utils-20170307-1-upgrade-requires-manual-intervention/).

```sh
sudo pacman -Syuw
sudo rm /etc/ssl/certs/ca-certificates.crt
sudo pacman -Su
```

Now you can install developer stuff.
```sh
sudo pacman -S --needed --noconfirm base-devel linux-headers git
```

After that you can try again to install both johnny e raspi-io

I should probably put this stuff in the paper page.

However, this ticket was about connecting via SSH and I've no idea why we're discussing these things in here.

Hard to find for others that might have the same problem :disappointed:

maertz, 2017-06-30:
Thanks for the support, I'll start over with a fresh install and try to make it work.
Sure I'll create a new ticket to describe the issue I'm having with my RPI3 and maybe send a PR to update the docs :)