WebReflection/benja #9: Wifi and additional questions

Stan92, 2017-04-17:
First of all, I thank you for sharing..
I try to set up the wifi for the PI but without success using wifi-menu
My additional questions, using Benja with a Raspberry, could I access to : 
- The PI Cam?
- The Bluetooth?

I m sorry if the questions seem stupid, I m very new to Linux but Benja is really what I'm looking for..
Thanks once again..

WebReflection, 2017-04-17:
wifi-menu , once it works, requires to be enabled through `netctl enable your-wifi-name` or it won't connect on each boot.

If you need to use the benja terminal, simply rename `package.json` in `package.jsoff` inside the app folder.

Benja will start as terminal and you could do whatever you need directly from there, including running wifi-menu, which should just work.

The PI Cam can be accessed through any software that access it,  but I'm afraid there's no direct bridge between the camera and the Electron app,  you need to eventually make one.

Same goes for the bluetooth, you can do anything you could do through your NodeJS side of the application, or simply spawning processes through child_nodes.

closed 2017-04-17 by WebReflection

LoganArnett, 2017-06-30:
Any recommendations on connecting to wifi other than `wifi-menu`? I cant seem to connect when using Benja at all

WebReflection, 2017-06-30:
I could use wifi-menu without issues but for any ArchLinux related question, ask ArchLinux ARM related forum.

Benja is **just ArchLinux** , it's nothing different