WebReflection/benja #10: Support for ARM64

gowrav, 2017-04-23:
Support for 
> Odroid C2
> Nano Pi K2

WebReflection, 2017-04-23:
The problem is Electron, AFAIK not built for ARM64 yet, otherwise it'd be there already

gowrav, 2017-04-27:
Thanks... lets keep this open for a resolve..

WebReflection, 2017-04-27:
Nothing I can resolve here though.

zeke, 2017-06-14:
👋 I'm on the Electron team. We are hoping to have an ARM64 build soon. I will try to remember to report back here when we do.

WebReflection, 2017-06-29:
@zeke did you forget to report? :smiley: 

just asking for a possible ETA so I can either close or wait and publish a new ISO for Pi3, thanks

zeke, 2017-06-29:
> did you forget to report?

Didn't forget. It just doesn't exist yet.

WebReflection, 2017-06-29:
OK, closing for maintenance sake but please do notify me/us/here if there's any progress, thank you!

closed 2017-06-29 by WebReflection