WebReflection/benja #12: Raspberry Pi3 Full HD performance

osmanaktas, 2017-05-02:
I tested some apps on Raspberry pi 3. It working nicely with normal applications. But there are quality problems in high quality videos (1920x1080). Also when I want to rotate the screen 90 degrees. The system is crashing.

Do you have  optimization recommendations or sample configuration for video applications Rpi 3 config.txt ? 

Thanks?  Sorry for my poor english :)

WebReflection, 2017-05-02:
the optimal screen resolution for the raspberry pi is 800x480, or at least that's the official display resolution and everything I've tried in there looks good.

The pi3 here is still not fully at its potentials because there's no ARMv8 ArchLinux image yet and on top of that, not any official Electron for ARMv8 so Benja would never work there.

I haven't had any problems rotating the screen through the config, but I haven't tried at 1080p (it shouldn't matter, really).

About video applications, I'm not sure what you mean. Benja is about IoT related projects based on Electron and NodeJS, if you mean playing videos only I think you should consider falling back to a Weston and HW Accelerated software for video only via Gtk or Qt and drop ELectron if FPS are essentials.

Consider that Kodi runs discretely well on Rpi3 at 1080p so I guess it can go faster with the right underlying software.

maxpowel, 2017-06-08:
My rpi3 also crashes when playing hd videos at 1920x1080

WebReflection, 2017-06-29:
Closing for maintenance sake.

closed 2017-06-29 by WebReflection