WebReflection/benja #13: Raspberry Pi 2 System Update fails whole benja startup

TheAifam5, 2017-05-22:
Hey,
Yesterday i used official ISO from benja.io and I tried to update the system.
The first ask you about mesa package that is replaced by other package.
So i successfully upgraded system but after the restart doesn't work anymore.
I tried also install everything manually but i seen in logs that electron can't start with xinit.

Regards TheAifam5

WebReflection, 2017-05-22:
are you sure you tested the latest ISO? I have changed it to use Xorg and native VC4 mesa instead of the old weston-pi blob based distro, maybe you should try to re-download the current ISO for Pi 2/3 and try to update that one instead of the old one.

If that's what you've done already, I'll have a look, thanks.

TheAifam5, 2017-05-23:
Hey,

So i lost like 8hrs of my day to figure out what exactly makes my issue.

After full update Xorg doesn't work with official Raspberry Pi Touchscreen 7" but via HDMI works. I found that the package "linux-firmware" makes problems, exactly "vc4-kms-v3d". After changing to the experimental version ""vc4-fkms-v3d" touchscreen works.

WebReflection, 2017-05-23:
gosh, thanks for investigating. what do you think I should do? I'd rather stick with mainstream but this is bad

TheAifam5, 2017-05-23:
So i will report that to the mainstream rpi2 firmware repo and if that is a bug on their side, u can apply small patch like i wrote before (just change "vc4-kms-v3d" to "vc4-**f**kms-v3d").

WebReflection, 2017-06-29:
did 4.9 work ?

WebReflection, 2017-07-18:
FYI I've fully updated the Raspberry Pi 2/3 iso. It now uses Weston 2, vc4-fkms-v3d (works with both HDMI and RPi Display), node 8.X and npm 5.x + Electron 1.6.x

closed 2017-07-18 by WebReflection