WebReflection/benja #15: Raspberry Pi Zero (W) support?

zeke, 2017-06-14:
Hi @WebReflection! Do you know if benja can be installed on a Zero?

WebReflection, 2017-06-14:
well, it's again a matter of Electron builds. ArchLinux and VideoCore 4 works pretty well there:
https://www.youtube.com/watch?v=hCLhGqadv04

TL;DR Gtk+ and WebKitGTK+ works like a charm there but the architecture is not ARMv7 but ARMv6 (or ARMv11) meaning current Electron ARM builds won't work and to be honest, if I remember correctly, there's no way to build Chromium on Raspberry Pi 1 or Pi 0 (same CPU) so I am not sure you want to go down that path but it'd be awesome!

zeke, 2017-06-14:
> there's no way to build Chromium on Raspberry Pi 1 or Pi 0

Do you know why that is?

WebReflection, 2017-06-14:
I told you, different ARM architecture.

However, at some point Googlers said they would have brought Chromium to Pi 1/0 but that was long time ago and afaik nothing happened

WebReflection, 2017-06-29:
Nothing happened here, closing for maintenance sake.

closed 2017-06-29 by WebReflection