WebReflection/benja #16: RPi2 doesn't boot at all

footniko, 2017-06-18:
Hi,
First of all, thank you for a great software!
I'm having issues running BENJA on my RPi2. I did all docs describes but when I put the card to the board, nothing happens - both green and red lights stay without blinking and no image. I've tried with 2 different microSD cards (both 10 class/8GB).
Any ideas?

footniko, 2017-06-19:
Just tried with Odroid C1 and it works perfectly. Any ideas why it doesn't work on RPi2?

WebReflection, 2017-06-19:
I have no idea because:

  * I don't know which system you are using
  * I don't know how you are flushing the sd card
  * I don't know if your raspberry Pi is connected to some screen that does not provide touch/mouse interface so it might not be able to start if you are using the image based on Weston
  * I don't know which image file you are using

footniko, 2017-06-19:
@WebReflection, thanks for the fast response!

- I'm using ubuntu 16.04 and flushing the card with `dd` command just like described in the docs. I tried the same on Odroid-C1 and it works fine, so on this step there is no fault.
- I tried on both Waveshare 10.1 touch screen (which works great on RPi2 with Ubuntu mate) and simple 17 inch display.
- I'm using this image file: https://s3.amazonaws.com/benja.io/iso/benja-rpi2.zip

I just tried to install BENJA from the scratch like docs describes and here what I had during the installation:
```
  using ArchLinuxARM-rpi-2-latest.tar.gz
  
  verifying /dev/sdb
  SD card total space: 7864320
  creating a root partition of 6000M

 [WARNING]
  disk /dev/sdb will be completely erased
  and every data in it will be lost.

  prepare Raspberry Pi 2 and erase /dev/sdb? [y/N]y  downloading ArchLinuxARM-rpi-2-latest.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  304M  100  304M    0     0  2264k      0  0:02:17  0:02:17 --:--:-- 2053k
  would like to save ArchLinuxARM-rpi-2-latest.tar.gz into ~/Downloads? [Y/n]Y
  partitioning

OK
  formatting
mke2fs 1.42.13 (17-May-2015)
  preparing first boot
  extracting the OS (this might take a while)
sudo: bsdtar: command not found
mv: cannot stat 'root/boot/*': No such file or directory
  moving prepared files
mv: cannot move 'BENJA-disk.service' to 'root/etc/systemd/system/BENJA-disk.service': No such file or directory
mv: cannot move 'BENJA-platform' to 'root/root/BENJA-platform': No such file or directory
mv: target 'root/root/' is not a directory
chmod: cannot access 'root/root/install': No such file or directory
chmod: cannot access 'root/root/wifi-setup': No such file or directory
sh: 1: cannot create root/etc/systemd/logind.conf: Directory nonexistent
sh: 1: cannot create root/etc/systemd/logind.conf: Directory nonexistent
  cleaning up

Please remove the SD card and boot it via Raspberry Pi 2.
Once logged in as user root with password root
simply write ./install and wait for it to reboot.

Please note that a network connection is necessary
and the first intallation might  take a while.

Thanks for using BENJA

```

After manual installation I got the same result - system doesn't boot at all, it seems like it doesn't even recognize the microSD, because it behaves the same like it was without the card.

WebReflection, 2017-06-19:
the whole system is ArchLinux based, I don't think you can have it working out of the box in Ubuntu or at least not until you have `bsdtar` errors in there.

You have two options:

  * try to install bsdtar and every other possible dependency that might be needed (dunno 'cause Ubuntu has different libraries names than ArchLinux) and do the manual procedure
  * attach a USB mouse to the RPi 2 and see if once you do that it boots

The last thing I am thinking might cause issues is the boot configuration, something you can clean up of the HDMI part which might cause issues indeed.

Last, but not least, putting the SD card in your computer, going into the APP folder, running npm install and npm start should work.

If that does, we might try to figure out which part of the image is having issues.

footniko, 2017-06-19:
Just installed `bsdtar` and tried to install from scratch again. This time without issues. However behavior is the same - seems like RPi doesn't even try to boot the image, I don't even have current that should be provided to my USB/ethernet after I plug the board to the rosette.
I think there should be something with my MicroSD card or Rpi2. I'll try to install arch linux and will let you know, thanks!

footniko, 2017-06-19:
@WebReflection, as I suspected the problem was in my microSD cards. I found original SD-microSD adapter to my Sandisk card and now everything works fine.
Thank you for your attention and for a great product! :)

closed 2017-06-19 by footniko