WebReflection/benja #17: Ability to run packaged application

footniko, 2017-06-19:
Actually it's not an issue, I'm just trying to find the way to run packaged electron application with BENJA.
By default BENJA starts the app by invoking `npm start`. How can I change it just to start packaged application?

WebReflection, 2017-06-19:
What do you mean?

WebReflection, 2017-06-19:
I mean ...you know you can just change the start script inside package.json, right?

footniko, 2017-06-19:
Sure :) 
I'm using a lot of libraries and use ES6 on my project. In order to not to use node compilers like babel, gulp, webpack etc., I'm using https://github.com/electron/electron-compile to compile my code on the fly. But I can't do this on production, so for production I use https://github.com/electron-userland/electron-packager
Electron-packager just package whole application to one executable file (let's say `app`) which I can run just by invoking `./app`

WebReflection, 2017-06-19:
and what has any of this to do with benja?

this is the `package.json`
https://github.com/WebReflection/benja/blob/gh-pages/os/app/package.json

nothing you mentioned is needed or even installed, or better, electron is just there for you.

What you do with your own app it's your business, and all you have to do is to change this line
https://github.com/WebReflection/benja/blob/gh-pages/os/app/package.json#L7

into this one:
```js
"scripts": {
    "start": "./your-app"
  },
```

am I missing something?

WebReflection, 2017-06-19:
**Remember** if you want differentiate between your PC (dev) and the Pi you can:

```js
"start": "if [ '$(uname -m)' = 'x86_64' ]; then electron index.js; else ./your-app; fi"
```

WebReflection, 2017-06-19:
watch out, I've edited the check

closed 2017-06-19 by WebReflection