WebReflection/benja #20: Papirus install instructions

maertz, 2017-06-27:
I'm having an issue with these papirus instructions.
https://github.com/WebReflection/benja/tree/gh-pages/examples/papirus

```
# install dependencies
sudo pacman -Syu --needed --noconfirm python-imaging fuse bc git i2c-tools ttf-freefont ntp
```
```
error: failed to commit transaction (conflicting files)
ca-certificates-utils: /etc/ssl/certs/ca-certificates.crt exists in filesystem
```
```
# use the Papirus installer, ignore apt-get warnings
curl -sSL https://goo.gl/i1Imel | sudo bash
```
```
bash: line 4: apt-get: command not found
bash: line 5: apt-get: command not found
bash: line 6: apt-get: command not found
bash: line 7: apt-get: command not found
bash: line 10: raspi-config: command not found
bash: line 11: raspi-config: command not found
```

Do you've an idea? Actually I just want to install raspi-io but 

```
[benja@benja ~]$ cd app/
[benja@benja app]$ npm install raspi-io --save
```

unfortunately results in 
```
[benja@benja ~]$ 
[benja@benja ~]$ cd app/
[benja@benja app]$ npm install raspi-io --save

> raspi-gpio@4.0.2 preinstall /home/benja/app/node_modules/.staging/raspi-gpio-b07dd7d7
> sh ./script/pigpio.sh

pigpio doesn't appear to be installed, installing now. You may be asked for your password.
[sudo] password for benja: 
sudo: apt-get: command not found

> raspi-pwm@4.0.2 preinstall /home/benja/app/node_modules/.staging/raspi-pwm-f26a761c
> sh ./script/pigpio.sh

pigpio doesn't appear to be installed, installing now. You may be asked for your password.
sudo: apt-get: command not found

> raspi-soft-pwm@3.0.0 preinstall /home/benja/app/node_modules/.staging/raspi-soft-pwm-d5cb7d4e
> sh ./script/pigpio.sh

pigpio doesn't appear to be installed, installing now. You may be asked for your password.
sudo: apt-get: command not found
app@1.0.1 /home/benja/app
├── raspi@4.0.0  extraneous
├── raspi-i2c@5.0.4  extraneous
├── raspi-io-core@2.0.1  extraneous
├── raspi-led@1.5.5  extraneous
└── raspi-serial@3.1.0  extraneous

npm WARN app@1.0.1 No repository field.
npm ERR! Linux 4.4.45-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "raspi-io" "--save"
npm ERR! node v7.5.0
npm ERR! npm  v4.1.2
npm ERR! code ELIFECYCLE

npm ERR! raspi-gpio@4.0.2 preinstall: `sh ./script/pigpio.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the raspi-gpio@4.0.2 preinstall script 'sh ./script/pigpio.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the raspi-gpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh ./script/pigpio.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs raspi-gpio
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls raspi-gpio
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/benja/app/npm-debug.log
[benja@benja app]$ 
```

Tried to install pigpio independently in front but there is no "apt-get" :/

WebReflection, 2017-06-27:
this is archlinux, no debian or ubuntu.

the issue you are having seems to be relatewd to this one:
https://www.archlinux.org/news/ca-certificates-utils-20170307-1-upgrade-requires-manual-intervention/

I need to either update the image or the documentation.

Thanks.

maertz, 2017-06-27:
> this is archlinux, no debian or ubuntu.

I guess that was related to "apt-get", sure was just mentioning because the link in following line contains apt-get commands.

`curl -sSL https://goo.gl/i1Imel | sudo bash`

Thanks for the quick response, i'll try the work around to update at least the os - thanks for pointing it out.

Any idea regarding the raspi-io installation or it's just me doing smth wrong?

```
[benja@benja ~]$ cd app/
[benja@benja app]$ npm install raspi-io --save
```

WebReflection, 2017-06-27:
if it's about warnings, it says: _use the Papirus installer, **ignore apt-get warnings**_
I believe I've written that for a reason

About the rest, please try to solve one thing per time. Update all the things after solving the credentials issue, then we'll see the rest :wink:

maertz, 2017-06-29:
Sorry was so excited - had to run all at the same time :D

closed 2017-06-29 by maertz