WebReflection/benja #21: Boot Time

LoganArnett, 2017-06-29:
Really enjoying Benja so far but I was curious if you had any ideas on how to possibly reduce the boot/load time when turning the Pi on? Right now it takes about 30-40 seconds most times to get up and running and to show the loading screen. Just trying to see if there is any way to speed this up at all

WebReflection, 2017-06-29:
npm is slow on ARM, deadly slow. Put Electron on top, there you have your boot time.

It's a pity, because ArchLinux is one of the fastest booting distro I've ever tried/tested/used.

If you build the GTK+ version which spins WebKitGTK instead and ditches Electron, you'll count no more than 20.

That being said, booting a Pi shouldn't be a common operation, rather a one off.

Maybe I'll figure out how to show astatic image till it works so that everyone can put their logo in there ... however, I'm focusing on other projects these days, and Benja never gained the amount of attention I hoped to keep pushing harder to make it optimal, sorry.

closed 2017-06-29 by WebReflection

LoganArnett, 2017-06-30:
@WebReflection Would you ever consider open sourcing the actual Benja source? I'm definitely interested in taking a look at continuing the project or even just looking at adding a simple logo/loading video while the application loads

WebReflection, 2017-06-30:
It's already Open Source .... it's entirely in this repository!!!

WebReflection, 2017-06-30:
https://github.com/WebReflection/benja/tree/gh-pages#how-to-install-benja-environment-from-the-scratch---arm

WebReflection, 2017-06-30:
The folder with the install and every other utility/helper for all supported platforms is here:
https://github.com/WebReflection/benja/tree/gh-pages/os

LoganArnett, 2017-06-30:
Oh awesome! I didn't see that, thanks!