WebReflection/benja #24: What HW acceleration is performed?

gamelaster, 2017-07-23:
Hello,
I want to ask, how is Electron HW accelerated?

Thanks

WebReflection, 2017-07-23:
in which platform?

gamelaster, 2017-07-23:
Raspberry Pi for sample

WebReflection, 2017-07-23:
latest ISO uses XWayland on Weston 2 so you should have HW acceleration through VC4 GPU, latest Mesa drivers, and the `vc4-fkms-v3d` dtoverlay. 

https://github.com/WebReflection/benja/blob/gh-pages/os/sh/utils-platform/rpi2#L122-L124

gamelaster, 2017-07-23:
Oh, thank you a lot. One more question, how it's made for render Electron without X server?

WebReflection, 2017-07-23:
that's XWayland on Weston 2. Unfortunately I still need to pass through XWayland because Chromium uses X as target backend and there are no binaries on AUR or ArchLinux using ozone (or at least none updated and building [this](https://github.com/Igalia/chromium) on Pi would take weeks I guess).

gamelaster, 2017-07-23:
It's not possible to compile it via cross-compile? I cross-compiled FFMPEG without any problems in x86_64 to ARMv7(L)

WebReflection, 2017-07-23:
it is, of course, but I don't have resources to do that on every Chromium update and the Benja installer is performed directly on the running hardware.

Moreover, once you have Chromium on Ozone, you have to hook that into Electron and I've no idea if you could cross compile directly Electron for the Pi through Ozone (Wayland) version.

If you need just a Wayland Web kiosk, without Electron, you can go right away with WebKit2GTK and start the a full screen browser instead:
https://github.com/WebReflection/archibold.io/blob/gh-pages/test/gjs/browser

WebReflection, 2017-07-23:
do you think we can close this?

skerit, 2017-08-01:
So everything is enabled, from WebGL to video decoding, @WebReflection?

I'm currently thinking about using either Benja or WPE for my project. An integrated node.js/electron is a definite plus, but I saw some very nice videos of WPE in action (and it renders directly to the framebuffer, I believe). Is there something else Benja does better than WPE (or worse :))?

WebReflection, 2017-08-01:
What's WPE ?

skerit, 2017-08-01:
It's the ["Web Platform for Embedded" by Igalia](https://www.igalia.com/wpe/)

Here's an implementation (and some gifs of) running on a resion.io image:
https://github.com/resin-io-playground/resin-wpe

WebReflection, 2017-08-01:
that looks like Benja bootstrapping into WebKitGTK+ instead of Electron, which is basically this:
https://www.youtube.com/watch?v=hCLhGqadv04

based on this folder:
https://github.com/WebReflection/benja/tree/gh-pages/os/jsgtk

to be honest, a basic ArchLinux that boots into Weston and WebKitGTK+ would be my preferred choice, specially because WebKitGTK has superior ECMAScript support, including native asynchronous modules via `<script type="module">`.

The bad part is that there is no electron integrated, so the communication channel between the browser and the backend would be clunky.

I've experimented already with few ways to improve this part but none of them was too convincing.

This was one successful, and dangerous, attempt:
https://github.com/WebReflection/trojan-horse

gamelaster, 2017-08-01:
I don't have any more questions, so when discussions ends, close it. Thank you 😊

WebReflection, 2017-08-01:
so you decided for WPE at the end?

I'm working right now to a NodeWorker solution. It's like Web Workers, but it runs on nodejs and it communicates through sockets.

I think I'll give this a try, it looks like it could be the solution to every problem and it could work with any browser.

WebReflection, 2017-08-01:
P.S. here my freshly baked alternative to trojan-horse: https://github.com/WebReflection/node-worker

skerit, 2017-08-02:
Are you thinking about switching Benja over to use `node-worker`? Or maybe offering multiple frontends the user can choose from?

I finally compiled the WPE buildroot and run it on my pi, and it does what it promised. Accelerated video and all. They actually forked Webkit and made lots of changes. While reading [their commits for their Webkit fork](https://github.com/WebPlatformForEmbedded/WPEWebKit/commits/master) it seems like they actually made lots of improvements. Some of them even show up in the original Webkit repository, so they are upstreaming some of their changes.

I do wonder if using a regular WebkitGTK or QtWebkit based browser would be as good, or if WPE can even run on anything else than their customized image.

WebReflection, 2017-08-02:
> I finally compiled the WPE buildroot and ...

One of the goals of this project was to provide a working ISO out of the box.

Using ArchLinux grants you're on latest/greatest software builds.
Using Wayland via Weston grants lighter software and HW acceleration where possible.

The Pi has great Mesa support and it's getting better so that offering an easy to maintain and update distro like ArchLinux, bootstrapping on a thin GJS WebKitGTK+ layer would be ideal.

Benja can already do that, but when it does it loses the `E` in its name.

`node-worker` was born yesterday evening, and only today I can see if it can provide everything I need.

**TL;DR**

Today I'll try to create from scratch an ISO that boots via GJS into WebKitGTK on Weston **and** makes everything that failed in [this other bug](https://github.com/WebReflection/benja/issues/22) work.

If I manage to do so, I'll think what do about this project.

The fact Electron diverges in built modules and NodeJS version is some sort of un-maintainable nightmare I don't have resources or time to deal with ( my patreon keeps getting ignored :smile: ).

Anyway ... let's see what I can do

skerit, 2017-08-02:
Oh yeah, I wasn't implying switching over to the WPE image, more like just using their version of the browser ON arch. WPE or WebKitGTK+, I'm curious to see how it goes either way :)

marqueeeeeee, 2017-08-15:
Hi im curious about this 
https://github.com/WebReflection/archibold.io/tree/gh-pages browser example. how can i run it inside benja? thank you ! I'd like to do some experiments with browsers that is enabled with HW.

WebReflection, 2017-08-15:
@marqueeeeeee you can try this folder as APP and see what happens:
https://github.com/WebReflection/benja/tree/gh-pages/os/gjs

However, Electron on Pi is already HW accelerated **but** the latest QML version is the best:
https://github.com/WebReflection/benja/issues/22#issuecomment-320229837

*But** it uses [node-worker](https://medium.com/@WebReflection/nodeworker-an-electron-alternative-for-the-internet-of-things-1eb64440d1e5) instead of Electron to talk to the backend

marqueeeeeee, 2017-08-16:
i'll try it tonight :D thanks! this is a really cool project. My goal is to make a digital signage in my room for dashboard purposes and media player/streamer.

WebReflection, 2017-09-22:
I guess it worked.

closed 2017-09-22 by WebReflection