WebReflection/benja #26: Cylon and Splash Screen

facetime88, 2017-08-05:
Hi,

I am new with RPI so I am sorry if these question sound silly.

I am working for a web app that use Cylon to connect to an arduino board. Can I install and use Cylon instead of Johnny Five?

Another question, can I change the splash screen?

Thank you.

WebReflection, 2017-08-05:
You can try installing on the Pi globally:

`npm install -g cylon cylon-firmata cylon-gpio cylon-i2c`

It should work. You can also remove johnny-five (globally) at that point.

> can I change the splash screen?

There is **no** splash screen. Benja boots to Electron (or latest prototype to QtWebEngine)

What you show there once the WebView starts is up to you.

facetime88, 2017-08-07:
Awesome....Thank you...!

facetime88, 2017-08-07:
I have another question: with the OS will boot to the electron app, how do I shutdown the RPI safely? Is it safe just plug off the power cable?

WebReflection, 2017-08-07:
you can spawn a `'shutdown', ['-h', 'now']` call and expose it through the Web interface **or** you can ssh into the RPi and do `sudo shitdown -h now` there.

You have **full** control of your system

closed 2017-08-07 by WebReflection