WebReflection/benja #27: How to get start writing app

facetime88, 2017-08-07:
Hi,

Again so sorry if this sounds silly.
I have flash the image and it boot successfully showing "Benja" landing page.
Now I'd like to create an app. What is the best method for development?

Currently I work in Windows, so I pull out the micro sd and pull it in to my windows PC. I browse the micro sd, but I am not familiar with all the files. There is only 1 folder named "overlay" and the rest is file with extension I don't familiar with. The micro sd disk size also significantly decrease to only about 90MB.

My question is, where my web app files need to placed? Is it in the micro sd or 'somewhere' in the rpi board embedded memory (if any)? What files structure it will looks like? Why the disk size is only 90MB, does it mean the app files is limited to no more than 90MB?

Thank you.

WebReflection, 2017-08-07:
When you plug the card in Windows (or any other OS) you'll have a BENJA disk (or APP) containig `package.json` and other files.

With that folder, you can access it via terminal and do `npm install`.

After that, you can `npm start`, assuming you have electron installed globally already.

If not, you can `npm install -g electron` and then try `npm start` again.

The folder you are seeing is just the boot one, and you should never touch it unless you know what you are doing.

Look closer, and pick the other one, the one that looks like a classic npm project.

WebReflection, 2017-08-07:
http://benja.io/docs/#how-to-develop

closed 2017-08-07 by WebReflection

facetime88, 2017-08-07:
I use 16GB micro sd.
There is only 1 partition shown in the explorer for my micro sd, which is the 94MB size.

After I check with windows disk management, It shows that there are another 3 partitions in the disk, so the total is 4 partitions, as shown in attachment.
[https://ibb.co/g7tP6a](url)

The second and the third doesn't have any file system, and the fourth is still unallocated.
Am I wrongly flash the disk? Or there is some steps I am missing here?

Thank you.

WebReflection, 2017-08-07:
If the project works on the Pi it means you flashed correctly **but** the ISO was created on an 8GB SD card.

There are 3 partitions + 1

  * the boot, which is the first about 100M
  * the system, an ext4 of 6GB (you don't have/need to see it)
  * the BENJA / APP which is another fat partition and you can extend that up to the whole free space you have, which is the unallocated space

I don't know why your Windows is not showing you the benja folder, it does on Linux and on Mac, and the partition is a FAT32 one.

Which Windows are you using?

facetime88, 2017-08-07:
I am using Windows x64.

Hmm....it is weird. I just try remotely access using SSH and type npm, and it works. So it means the pi recognized the partitions and all seems works fine in Pi. But why the windows doesn't recognize it I do not know.

I see that the boot partition is FAT not FAT32, is that okay?

WebReflection, 2017-08-07:
Again ... the system works on PI and it boots to the landing / demo page.

Windows does not understand partitions like other operating system, it has its own (like other operating systems) but the APP partition is FAT (doesn't matter how you see it, as long as it's FAT).

Windows is a second class citizen when it comes to IoT projects so I haven't tested much in there but benja is supposed to be usable in there too so I'll check what could be the issue, if I find a windows around.

facetime88, 2017-08-07:
:D Yeah.....I really feel sucks has been bound to Windows for the last 15 years.
Ok, I'll try it in 8GB instead. I'll let you know if that make difference.

Thanks.

WebReflection, 2017-08-07:
Gotta be honest with you, the only thing Windows does better than others is Gaming because they keep developing drivers and games mostly for Windows.

Everything else, when it comes to power/user or programming experience, is tyring to catch up but far away from Linux (which meanwhile became the best Desktop OS of them all).

I'm flushing now and I've found a Windows 10 x64, I should be able to try there soon.

WebReflection, 2017-08-07:
[image]

WebReflection, 2017-08-07:
I've just plugged in the USB stick and everything went fine.

Have you tried using [Etcher](https://etcher.io/) to burn the ISO ?

facetime88, 2017-08-07:
This...so weird...what's wrong with my windows...:(

Yes.. I am using Etcher, and just tried it again using win32imager, still the same result.
Then I install a free non-windows partition manager. And it shows the files system is FAT32 for BENJA-APP partition. But no drive letter assigned yet. I assigned it a drive letter and still not showing.

I am now re-flashing again using Etcher, and after completing assign drive letter, I will try to give it a computer restart. I will let you know soon. Still flashing now..

WebReflection, 2017-08-07:
I'm afraid beside trying on a "virgin" (fresh new install) Win10 Home 64x system there's nothing else I can do. I haven't used anything special at all. There's no external software neither ... literally nothing from base Win10.

facetime88, 2017-08-07:
[image]

Okay, after restarting my windows, still the same. I think the problem is because the second partition file is not recognized by windows (set as OTHER). This make the next partition, which is BENJA APP and unallocated space can't be recognized. I tried to create a new partition from unallocated space and it says the action can't be done since windows just recognized the first partition.

May I know what is the file system for the second partition (5.5GB) ? I see at your side it is recognized by windows, so it must be whether FAT, FAT32, or NTFS. But why on my side is unknown (set as OTHER), I can't understand.

I will try to buy a 8GB instead tomorrow, maybe it will make difference.

WebReflection, 2017-08-07:
> I think the problem is because the second partition file is not recognized by windows (set as OTHER).
> This make the next partition, which is BENJA APP and unallocated space can't be recognized. 

did you forget it just works on my Win10 Home ?

> May I know what is the file system for the second partition (5.5GB) ?

Ext4

> I see at your side it is recognized by windows

Not at all, it says it doesn't recognize it and it asks me to format it, which I say no and ignore happily ever after.

Which version of Windows are you running? Please don't tell me just the architecture, thanks.

facetime88, 2017-08-08:
Wooahh... it works for you and it should work for me also. 
I am using Windows 10 Prof x64.

I will try it again after I get a brand new 8GB micro sd today.

facetime88, 2017-08-08:
I just found out that the free mini tool partition wizard recognized and offer format in ext4 filesystem. But the partition made by iso image is not recognized / set as "other".

WebReflection, 2017-08-08:
You **don't need** to recognize Ext4

facetime88, 2017-08-08:
Hi,

Just now I am trying again, but this time with my laptop Windows 10 Home 32 with previous micro sd, which is Sandisk 16GB Class 10. But the result still the same. So this is really weird. The problem seems definately not the Windows. Whether the problem is on the sd card, or I am missing some steps here.
But since the sd-card works perfectly in pi, so I can skip that the fault is on me.

What micro sd brand are you using? I'd like to try to use exactly what you have used.

WebReflection, 2017-08-08:
Any brand would do. You have an SD card with 3 partitions + empty space.

First partition is FAT16 for boot. You don't need to interact with it.

Secondo partition is a Linux Ext4 for the OS. You don't need to interact with it.

Third partition is a FAT32 that automount in Linux, macOS, OSX, and Windows 10 (I've tried 2 laptops and both are just fine).

I start believing the issue is in the burner tool you are using. Maybe it does something weird on Windows only so that if you burn the iso via `dd` from a Mac or a Linux PC you won't have the same issue.

Is this a possible test you can do? No, you cannot use VMs, you need a proper Linux / Mac machine.

**However** the SD I am using is a SanDisk one, which I recommend regardless to whoever wants to deal with SD cards. I've tried other brands though, and all of them just works.

What a weird situation, apologies I cannot help any further than this.

facetime88, 2017-08-09:
Okay, 

Just got a new 8GB sandisk micro sd. Same result :( I think you are right, there maybe a problem in the burner tool. So for now I am giving up working it in Windows. I setup mobaxterm and now working it remotely.

Thank's man for this OS. This help me a lot since I do not know Linux at all. So I can focus in developing the app.

facetime88, 2017-08-09:
Just let you know I just open a ticket in [Etcher](https://github.com/resin-io/etcher/issues/1682), hope someday you or somebody else can figure out what the cause of this issue and how to fix it.