WebReflection/benja #28: Permission Issue

facetime88, 2017-08-09:
I just tried to install Cylon and it shows me some errors as follow:

> [benja@benja app]$ npm install cylon-firmata cylon-gpio cylon-i2c
npm WARN vms@1.0.0 No repository field.
npm WARN The package electron is included as both a dev and production dependency.
npm ERR! path ../mkdirp/bin/cmd.js
npm ERR! code EPERM
npm ERR! errno -1
npm ERR! syscall symlink
npm ERR! Error: EPERM: operation not permitted, symlink '../mkdirp/bin/cmd.js' -> '/home/benja/app/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp'
npm ERR!  { Error: EPERM: operation not permitted, symlink '../mkdirp/bin/cmd.js' -> '/home/benja/app/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp'
npm ERR!   cause:
npm ERR!    { Error: EPERM: operation not permitted, symlink '../mkdirp/bin/cmd.js' -> '/home/benja/app/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp'
npm ERR!      errno: -1,
npm ERR!      code: 'EPERM',
npm ERR!      syscall: 'symlink',
npm ERR!      path: '../mkdirp/bin/cmd.js',
npm ERR!      dest: '/home/benja/app/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp' },
npm ERR!   stack: 'Error: EPERM: operation not permitted, symlink \'../mkdirp/bin/cmd.js\' -> \'/home/benja/app/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp\'',
npm ERR!   errno: -1,
npm ERR!   code: 'EPERM',
npm ERR!   syscall: 'symlink',
npm ERR!   path: '../mkdirp/bin/cmd.js',
npm ERR!   dest: '/home/benja/app/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp',
npm ERR!   parent: 'node-pre-gyp' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/benja/.npm/_logs/2017-08-09T08_35_00_300Z-debug.log

How to fix this? Many thanks.

WebReflection, 2017-08-09:
npm install **-g** cylon-firmata cylon-gpio cylon-i2c

WebReflection, 2017-08-09:
since you are on Electron version, you might want to follow these instructions, **skipping** the _Install PIGPIO and copy it's lib file to the user folder_ part:
https://github.com/WebReflection/benja/issues/22#issuecomment-319609051

WebReflection, 2017-08-09:
it's actually better this second way, if you install **-g** Electron might not be able to find those modules.

Try both

closed 2017-08-09 by WebReflection