WebReflection/benja #29: npm ERR! code ELIFECYCLE errno 1

facetime88, 2017-08-10:
I am facing an error as follow:

> npm verb lifecycle vms@1.0.0~start: unsafe-perm in lifecycle true
npm verb lifecycle vms@1.0.0~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/benja/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/benja/.npm-packages/bin
npm verb lifecycle vms@1.0.0~start: CWD: /home/benja/app
npm info lifecycle vms@1.0.0~start: Failed to exec start script
npm verb stack Error: vms@1.0.0 start: `electron index.js`
npm verb stack Exit status 1
npm verb stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:289:16)
npm verb stack     at emitTwo (events.js:125:13)
npm verb stack     at EventEmitter.emit (events.js:213:7)
npm verb stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
npm verb stack     at emitTwo (events.js:125:13)
npm verb stack     at ChildProcess.emit (events.js:213:7)
npm verb stack     at maybeClose (internal/child_process.js:897:16)
npm verb stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:208:5)
npm verb pkgid vms@1.0.0
npm verb cwd /home/benja/app
npm verb Linux 4.9.37-1-ARCH
npm verb argv "/usr/bin/node" "/usr/bin/npm" "start" "--verbose"
npm verb node v8.1.4
npm verb npm  v5.3.0
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! vms@1.0.0 start: `electron index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the vms@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm verb exit [ 1, true ]

Do you know what does it mean? Thank's bro.

WebReflection, 2017-08-10:
I don't know what you are doing so no.

I'm also very busy with another project these days, cannot play with the Pi much, sorry.

facetime88, 2017-08-10:
It's okay. Thanks for the great job of Benja.

closed 2017-08-11 by facetime88

WebReflection, 2017-08-11:
You closed this without any comment. Did you solve the issue?

Wouldn't be nice of you to share how for others that might have the same issue?

facetime88, 2017-08-12:
Sorry :)
This error came up usually as there is a type error in the package.json. So I fix it and it's just running fine.

WebReflection, 2017-08-13:
> there is a type error in the package.json

in which package.json there is an error?

facetime88, 2017-08-13:
The typing error is in my package.json app, not the built-in benja app 👍