WebReflection/benja #30: [Suggestion] `BENJA` environment variable

shroudedcode, 2017-08-22:
It would be great if Benja would always set a specific environment variable to true, maybe `BENJA` or `BENJA_OS`. This would allow developers to easily check whether their application is currently running on a Benja device, like this:
```javascript
if(process.env.BENJA)
  console.log("I'm on a Benja device! :D")
else
  console.log("I'm on a normal computer. :I")
```
[Travis CI](https://travis-ci.org/) offers [a similar variable called `TRAVIS`](https://docs.travis-ci.com/user/environment-variables/#Default-Environment-Variables) that indicates that a program is run by a Travis build.

WebReflection, 2017-08-22:
```js
if (/^arm/.test(process.arch)) {
  console.log("I'm on a Benja device! :D");
} else {
  console.log("I'm on a normal computer. :I");
}
```

closed 2017-08-22 by WebReflection

shroudedcode, 2017-08-22:
@WebReflection Thanks for the quick answer!

WebReflection, 2017-08-22:
sorry I rushed, but re-creating 12 ISO of 8 GB each to put some info you could discover otherwise is not my highest priority now.

However, you're totally right, there should be some unique env variable to check against.

That's the best way to go, but not these days. Apologies.

WebReflection, 2017-08-22:
P.S. will be in, next deploy!