WebReflection/benja #31: Doesn't booltload when modified

utiq, 2017-08-24:
I followed the steps to burn a new image and create a bootable SD and it works like a charm. I put the SD in the raspberry and it works fine.
I took the SD out and I put it in my computer to modify some of the code.
I just run:
npm install
npm start

The app worked, I didn't modify any code, but when I put back the SD in the Raspberry, it didn't boot. Just a black screen with a colorful line at the top.

Image I used: benja-rpi2.iso
My OS: MacOS Sierra
Raspberry: Raspberry Pi 3

WebReflection, 2017-08-27:
`npm install` brings in modules, including those that need native platform build steps.

Those cannot possibly be compatible with the raspberry PI and you need to remove `app/node_modules` so that the first time the Raspberry Pi will install only production modules (and **not** dependencies) and it will use its own pre-built, pre-compiled version of the dependencies from the global folder.

WebReflection, 2017-08-27:
TL;DR the flow in developing Benja apps locally, then put them on the Pi

  * `npm install` in your local machine
  * create the app, install and save only modules that don't need native builds
  * remove the node_modules folder before putting the SD Card in the Pi

Alternatively, to simplify your work flow:

  * `npm install -g native-modules`, where _native_modules_ are those that need build steps
  * `npm install --save js-modules`, where _js-modules_ are those that don't need node-gyp and other build steps
  * `npm install --production` to bring in only depenendencies that do not need node-gyp and other build steps

Once you have this setup, you don't need to remove the node_modules folder before putting back your sd card in the Rasperrby Pi, but you might need to install globally modules that requires native scripts via ssh (or other new modules too, the production one ... where again, you just drop the node_modules folder once and that's it)

closed 2017-08-27 by WebReflection