WebReflection/benja #32: VFAT filesystem for /home/benja/app prevents using app.makeSingleInstance()

rhclayto, 2017-08-25:
Since the `/home/benja/app` is formatted as VFAT, which cannot use UNIX sockets, it is impossible to use BENJA with Electron's `app.makeSingleInstance()` method, which depends on creating sockets in the app directory. Formatting as ext4 makes it possible to use this method. See https://github.com/electron/electron/issues/10349 .

rhclayto, 2017-08-25:
Another solution: change the TMPDIR environment variable in the main electron process to somewhere outside the app directory: 

```js
process.env['TMPDIR'] = '/tmp';
```

WebReflection, 2017-08-25:
awesome, so all you need to do is to change the package.json inside the VFAT folder to set such env variable upfront, right?

```
  "scripts": {
    "start": "TMPDIR=/tmp electron index.js"
  }, 
```

rhclayto, 2017-08-26:
Works for me!

closed 2017-08-26 by WebReflection

rhclayto, 2017-08-26:
You might want to add this information to the documentation.