WebReflection/benja #33: GPU Crashes with Wayland/Weston

rhclayto, 2017-09-02:
Hi, I'm experiencing GPU crashes with BENJA (on Raspberry Pi 3) when the process has been running for a number of hours. Usually, if I leave it overnight the GPU will have crashed by the morning & the HDMI monitor no longer receives signal from the Raspberry Pi. This is the relevant journalctl logs for when wayland/weston starts:

```
Sep 02 01:58:54 benja.local weston-launch[258]: Date: 2017-09-02 UTC
Sep 02 01:58:54 benja.local weston-launch[258]: [01:58:54.847] weston 3.0.0
Sep 02 01:58:54 benja.local weston-launch[258]:                http://wayland.freedesktop.org
Sep 02 01:58:54 benja.local weston-launch[258]:                Bug reports to: https://bugs.freedesktop.org/enter_bug.cgi?product=Wayland&component=weston&version=3.0.0
Sep 02 01:58:54 benja.local weston-launch[258]:                Build: unknown (not built from git or tarball)
Sep 02 01:58:54 benja.local weston-launch[258]: [01:58:54.847] Command line: /usr/bin/weston
Sep 02 01:58:54 benja.local weston-launch[258]: [01:58:54.847] OS: Linux, 4.9.43-1-ARCH, #1 SMP Fri Aug 18 01:10:29 UTC 2017, armv7l
Sep 02 01:59:11 benja.local kernel: Adding 239492k swap on /dev/zram1.  Priority:5 extents:1 across:239492k SSFS
Sep 02 01:59:11 benja.local kernel: Adding 239492k swap on /dev/zram2.  Priority:5 extents:1 across:239492k SSFS
Sep 02 01:59:11 benja.local kernel: Adding 239492k swap on /dev/zram3.  Priority:5 extents:1 across:239492k SSFS
Sep 02 01:59:11 benja.local systemd-timesyncd[242]: Synchronized to time server 172.104.8.139:123 (0.arch.pool.ntp.org).
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:54.847] warning: XDG_RUNTIME_DIR "/tmp" is not configured
Sep 02 01:59:11 benja.local weston-launch[258]: correctly.  Unix access mode must be 0700 (current mode is 777),
Sep 02 01:59:11 benja.local weston-launch[258]: and must be owned by the user (current owner is UID 0).
Sep 02 01:59:11 benja.local weston-launch[258]: Refer to your distribution on how to get it, or
Sep 02 01:59:11 benja.local weston-launch[258]: http://www.freedesktop.org/wiki/Specifications/basedir-spec
Sep 02 01:59:11 benja.local weston-launch[258]: on how to implement it.
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:54.918] Using config file '/home/benja/.config/weston.ini'
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:54.949] Output repaint window is 7 ms maximum.
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:54.986] Loading module '/usr/lib/libweston-3/drm-backend.so'
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:55.056] initializing drm backend
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:55.056] logind: not running in a systemd session
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:55.056] logind: cannot setup systemd-logind helper (-61), using legacy fallback
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:55.059] using /dev/dri/card0
Sep 02 01:59:11 benja.local weston-launch[258]: [01:58:55.066] Loading module '/usr/lib/libweston-3/gl-renderer.so'
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.890] EGL client extensions: EGL_EXT_platform_base
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_MESA_platform_gbm
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.891] warning: neither EGL_EXT_swap_buffers_with_damage or EGL_KHR_swap_buffers_with_damage is supported. Performance could be affected.
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.891] EGL_KHR_surfaceless_context available
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] EGL version: 1.4 (DRI2)
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] EGL vendor: Mesa Project
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] EGL client APIs: OpenGL OpenGL_ES
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] EGL extensions: EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_gl_texture_cubemap_image EGL_KHR_image
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_image_base EGL_KHR_image_pixmap
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_no_config_context EGL_KHR_reusable_sync
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_KHR_surfaceless_context EGL_KHR_wait_sync
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_MESA_configless_context EGL_MESA_drm_image
Sep 02 01:59:12 benja.local weston-launch[258]:                EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] GL version: OpenGL ES 2.0 Mesa 17.1.8
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] GLSL version: OpenGL ES GLSL ES 1.0.16
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] GL vendor: Broadcom
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] GL renderer: Gallium 0.4 on VC4 V3D 2.1
Sep 02 01:59:12 benja.local weston-launch[258]: [01:59:12.950] GL extensions: GL_EXT_blend_minmax GL_EXT_multi_draw_arrays
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_texture_format_BGRA8888
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_element_index_uint GL_OES_fbo_render_mipmap
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_EGL_image GL_OES_depth_texture
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_get_program_binary GL_APPLE_texture_max_level
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_discard_framebuffer GL_EXT_read_format_bgra
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_frag_depth GL_NV_fbo_color_attachments
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_EGL_image_external GL_OES_EGL_sync
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_OES_vertex_array_object GL_EXT_unpack_subimage
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_map_buffer_range GL_KHR_debug GL_OES_surfaceless_context
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_separate_shader_objects
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_compressed_ETC1_RGB8_sub_texture
Sep 02 01:59:12 benja.local weston-launch[258]:                GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp
[... 95 lines cut ...]
Sep 01 01:43:10 benja.local kernel: 7f60: 000000a8 0004092b b7206000 b7218181 b7218180 00000006 c0a06440 7eec7d00
Sep 01 01:43:10 benja.local kernel: 7f80: b7206000 00000000 00000040 8028fbdc 02969e60 7eec7d00 c0a06440 00000036
Sep 01 01:43:10 benja.local kernel: 7fa0: 80107c24 80107a80 02969e60 7eec7d00 00000006 c0a06440 7eec7d00 00000000
Sep 01 01:43:10 benja.local kernel: 7fc0: 02969e60 7eec7d00 c0a06440 00000036 7eec7d00 00000000 00000040 00000040
Sep 01 01:43:10 benja.local kernel: 7fe0: 76da9efc 7eec7cc4 76d90c28 7696c33c 60070010 00000006 00000000 00000000
Sep 01 01:43:10 benja.local kernel: [<7f40c450>] (validate_tile_binning_config [vc4]) from [<7f40cb50>] (vc4_validate_bin_cl+0xb0/0x2a4 [vc4])
[... cut ...] have spent all day wading through a welter of graphics-related stuff that didn't solve much of this problem, I thought I'd bring it up here.

A couple of questions, then. Is there some way first of all to fix this while still using Weston? For instance, a way to get weston to work on Raspberry Pi without using the experimental drivers?

Secondly, since everything is working OK with SDDM/Xorg, how are you accomplishing the touch reload reloading of the app? It no longer works with the SDDm setup.

rhclayto, 2017-09-02:
Also, after some testing, without the hardware acceleration, the scroll lag was just horrendous. Somehow I need to use the vc4 driver but without the crashing! It seems I'm in a bind. :-/

WebReflection, 2017-09-02:
would QtWebEngine be a valuable solution ?

It's HW accelerated on Xorg, but it needs node-worker instead of Electron.
https://github.com/WebReflection/benja/issues/22#issuecomment-320229837

rhclayto, 2017-09-02:
Investigating now.

rhclayto, 2017-09-04:
Well since switching back to `dtoverlay=vc4-fkms-v3d` I have had no crashes! Whereas before it used to be consistent, at least one crash within a 24-hour period. Perhaps I chcnaged something in the electron app itself that has corrected this, but if I did, I don't know what it is. So anyway, I am having no crashes with the following boot settings:

config.txt
```
gpu_mem=8
dispmanx_offline=1
dtoverlay=vc4-fkms-v3d
avoid_warnings=2
dtparam=audio=on
disable_overscan=1
```

cmdline.txt:

```
root=/dev/mmcblk0p2 rootfstype=ext4 rw rootwait quiet loglevel=0 splash console=ttyAMA0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 elevator=noop cma=256MB
```

So I'll close this.

closed 2017-09-04 by rhclayto

WebReflection, 2017-09-05:
oh gosh ... I use gpu_mem=256 as minimum, no idea how your Electron even starts with 8MB of GPU ... are you sure you need the GPU at all ?

rhclayto, 2017-09-05:
From what I understand, the open-source vc4 driver used by `dtoverlay=vc4-fkms-v3d` doesn't use `gpu_mem` at all.  Eric Anholt, who is writing the driver, has said things like this on several occasions in the GitHub issue queue: 'Note: The gpu mem setting in config.txt reserves memory that doesn't get used by the open graphics driver.' (https://github.com/anholt/linux/issues/28), '(Also note that the open driver doesn't want any gpu_mem)' (https://github.com/anholt/linux/issues/47), 'vc4-kms-v3d doesn't care about any of your config.txt display settings (except avoid_warnings=1 which is required). If any of config.txt did affect vc4-kms-v3d, that would actually be a bug. ' (https://github.com/anholt/linux/issues/92).

So it seems to open vc4 driver itself isn't responding to anything in config.txt. However, I can't boot up at all without something in `gpu_mem`, so perhaps it's needed for an early stage before the vc4 drivert akes over. I have `cma=256MB` in cmdline.txt, so RQAM should be assigned dynamically to the GPU as needed.

I am no expert at all in any of this, but what I have has been working for about three days. Fingers crossed it keeps working.