WebReflection/benja #35: Changing the ip and DNS to use SSH

carloslira, 2017-10-04:
Hi! Great work you've got there, I'm having a issue with SSH, can you guys help me out?

WebReflection, 2017-10-04:
"_you guys_" is just me :smiley: 

I don't have any board with me for the next 3 months (I'm in South America 'till mid January 2018) but if it's a simple ArchLinux config issue I might help regardless.

I need to know what is your issue though ...

carloslira, 2017-10-04:
Thanks for the answer, i just worked arround my previous problem, now I'm trying to get a 1920x1080 BrowserWindow, but I'm getting an black padding like the app doesn't fill the whole window, do you know what is going on?

WebReflection, 2017-10-05:
it's regular Raspberry Pi configuration thing. You need to set `disable_overscan=1` in the boot config.txt
https://elinux.org/RPiconfig

closed 2017-10-05 by WebReflection