WebReflection/benja #37: IP address not getting assigned for SSH

bmeeder22, 2017-10-25:
I followed along with the guide as posted in the docs. The app boots up no problem and I can click on the logos and see the screen size. The only problem however is that the IP address in order to connect to SSH is not present. I have ran ifconfig using Node and printed that to the screen and eth0 does not have an IP address listed. This should have been configured automatically so I am not sure what is going wrong. Note: I do have the ethernet cable connected at bootup and the lights are on. My computer however, does not show that the ethernet is connected until shortly after the app boots.

WebReflection, 2017-10-25:
* which SBC are you using?
  * is your router configured as DHCP?
  * are you sure the ethernet of your SBC works?

bmeeder22, 2017-10-26:
* I am using the Raspberry Pi 3 Model B
* I do not have a router between the RPi and the computer it is simply an ethernet cable connecting my RPi to my Macbook
* When I tested the board with normal Raspbian in a lab the ethernet worked and I could SSH into it

Things I am going to try and get back to you:
* I am going to switch out the power supply and see if that changes anything. There is a chance that the board isn't receiving enough power to operate the ethernet normally
* I am going to switch out the RPi to double check that ethernet works

Will test these today and get back to you on the results

WebReflection, 2017-10-26:
> When I tested the board with normal Raspbian in a lab

a "lab" is different than your MacBook ... you need a DHCP capable router. If your Mac can do that, then it's OK.

closed 2017-10-26 by WebReflection