WebReflection/benja #39: Issue installing npm package that must symlink

alanpilloud, 2017-11-08:
Hello,

First of all, thank you for your great work on Benja, it is really helpful !

I have an issue with my project which has some dependencies that generate an "Error: EPERM: operation not permitted, symlink xxx to yyy".

For testing purpose, I isolated one package : acorn.

This is the errorlog I get when running `npm i acorn --save-dev` 
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'i', 'acorn', '--save-dev' ]
2 info using npm@5.3.0
3 info using node@v8.1.4
4 verbose npm-session 898e80954f00baf4
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 304 https://registry.npmjs.org/acorn 591ms (from cache)
8 silly pacote tag manifest for acorn@latest fetched in 700ms
9 silly install loadIdealTree
10 silly install cloneCurrentTreeToIdealTree
11 silly install loadShrinkwrap
12 silly install loadAllDepsIntoIdealTree
13 silly resolveWithNewModule acorn@5.2.1 checking installable status
14 silly currentTree app@1.0.1
14 silly currentTree └── benja@0.1.1
15 silly idealTree app@1.0.1
15 silly idealTree ├── acorn@5.2.1
15 silly idealTree └── benja@0.1.1
16 silly install generateActionsToTake
17 silly diffTrees action count 1
18 silly diffTrees add acorn@5.2.1
19 silly decomposeActions action count 8
20 silly decomposeActions fetch acorn@5.2.1
21 silly decomposeActions extract acorn@5.2.1
22 silly decomposeActions preinstall acorn@5.2.1
23 silly decomposeActions build acorn@5.2.1
24 silly decomposeActions install acorn@5.2.1
25 silly decomposeActions postinstall acorn@5.2.1
26 silly decomposeActions finalize acorn@5.2.1
27 silly decomposeActions refresh-package-json acorn@5.2.1
28 silly install executeActions
29 silly doSerial global-install 8
30 verbose correctMkdir /home/benja/.npm/_locks correctMkdir not in flight; initializing
31 verbose lock using /home/benja/.npm/_locks/staging-7f39b336fab31ce0.lock for /home/benja/app/node_modules/.staging
32 silly doParallel extract 8
33 silly extract acorn@5.2.1
34 silly doReverseSerial unbuild 8
35 silly doSerial remove 8
36 silly doSerial move 8
37 silly doSerial finalize 8
38 silly finalize /home/benja/app/node_modules/acorn
39 silly doParallel refresh-package-json 8
40 silly refresh-package-json /home/benja/app/node_modules/acorn
41 silly doParallel preinstall 8
42 silly preinstall acorn@5.2.1
43 info lifecycle acorn@5.2.1~preinstall: acorn@5.2.1
44 silly doSerial build 8
45 silly build acorn@5.2.1
46 info linkStuff acorn@5.2.1
47 silly linkStuff acorn@5.2.1 has /home/benja/app/node_modules as its parent node_modules
48 verbose linkBins acorn@5.2.1
49 verbose linkBins [ { acorn: './bin/acorn' },
49 verbose linkBins   '/home/benja/app/node_modules/.bin',
49 verbose linkBins   false ]
50 verbose linkMans acorn@5.2.1
51 verbose unlock done using /home/benja/.npm/_locks/staging-7f39b336fab31ce0.lock for /home/benja/app/node_modules/.staging
52 warn app@1.0.1 No repository field.
53 verbose stack Error: EPERM: operation not permitted, symlink '../acorn/bin/acorn' -> '/home/benja/app/node_modules/.bin/acorn'
54 verbose cwd /home/benja/app
55 verbose Linux 4.9.37-1-ARCH
56 verbose argv "/usr/bin/node" "/usr/bin/npm" "i" "acorn" "--save-dev"
57 verbose node v8.1.4
58 verbose npm  v5.3.0
59 error path ../acorn/bin/acorn
60 error code EPERM
61 error errno -1
```
Here's what I've already tried : 

- Updating to npm 5.5.1
- Running `sudo npm i acorn --save-dev`
- Running `sudo npm i acorn --save-dev --unsafe-perm`

I assume that this is not a directly an issue with Benja, but I was hoping that you've already had the same issue.

Have a nice evening.

WebReflection, 2017-11-08:
have you tried `--no-bin-links` ?

alanpilloud, 2017-11-08:
Thanks, that worked !

closed 2017-11-08 by alanpilloud