WebReflection/benja #42: Switching to Xorg in RPi3

lahdekorpi, 2017-11-18:
@WebReflection

First of all, thanks a lot Andrea for all the hard work that's gone into the project! I've taken it as the base for an info screen project and been modifying the OS parts a lot.

One of the things I'm having problems with the Raspberry Pi 3 version is video playback performance. The playback is really laggy and dropping frames, both with the HTML5 video player inside Electron and just running Mplayer or VLC. Although these are straight from the repos and I haven't tried mplayer2 with `--enable-wayland`
Overclocking to the max doesn't change a thing, at all. And I get the same result with 3 different devices and memory cards.

TL;DR: Is there a way to force Xorg fallback to test if video playback performance is better?

PS: I've done a lot of improvements and fixes on the OS level, not the "sample" Benja app. Like logging, init scripts, crash handling, and a few minor security things. What is the correct way to make pull requests for these?

WebReflection, 2017-11-19:
best video playback is via Qt and there is an experimental ISO:
https://github.com/WebReflection/benja/issues/22#issuecomment-320229837

Electron is not fully HW accelerated but it uses already XOrg via XWayland

The Qt iso does not uses node.js on the browser but you can circumvent that via node-worker:
https://github.com/WebReflection/benja/issues/22#issuecomment-320330038

WebReflection, 2017-11-19:
P.S. about PRs , install-common or install-users might be surely improved here:
https://github.com/WebReflection/benja/tree/gh-pages/os/sh

if you have specific things for the pi only, this is the file:
https://github.com/WebReflection/benja/blob/gh-pages/os/sh/utils-platform/rpi2

lahdekorpi, 2017-11-19:
Thanks a lot for the quick response!
I wish I knew about the qml version before starting. I've actually already built many features on top of Electron's API so porting them to the worker model is going to take a bit of time. But that direction looks like a really promising one, so definitely going to try that. Will the project be renamed BQNJA? 😄 

For anyone else that might be reading this and have similar performance problems, but want to keep on using Electron, I've found a few things to get better performance:

1) Disable /etc/systemd/system/weston-compositor.service
2) Install `pacman -S xorg xorg-xinit xorg-server xterm`
3) Create your own service file for startx and use `.xinitrc` to load `/home/benja/.app-launcher` (or symlink)
4) Use omxplayer for videos if you can use local playback (node-omxplayer npm is great for this)

For some reason, I'm getting much better 2D and video playback performance by skipping Wayland.

WebReflection, 2018-02-22:
closing as there's no activity and not sure what I could do if not adding your latest hint somewhere in the documentation .

closed 2018-02-22 by WebReflection