WebReflection/benja #43: QEMU Emulation

mrefaie, 2017-11-19:
Is there any way to emulate BENJA using QEMU?

WebReflection, 2017-11-19:
you flush benja, you put in your computer, you type `npn install` and `npm start` ... that is the emulator?

lahdekorpi, 2017-11-21:
@mrefaie I just installed Arch Linux into a virtual machine as usual, and then manually installed Benja following the [instructions](https://benja.io/docs/#how-to-install-b-e-n-j-a-environment-from-the-scratch-i686-x86_64). This is only in the case you want to test the rest of the OS-level stuff, not just the app development as mentioned above.

WebReflection, 2017-11-21:
To install ArchLinux on a VM this might be handy: https://medium.freecodecamp.org/installing-arch-linux-from-scratch-b595095ddd48

WebReflection, 2018-02-22:
no answer in months

closed 2018-02-22 by WebReflection