WebReflection/benja #45: JSON.parse worker data?

lahdekorpi, 2017-11-20:
https://github.com/WebReflection/benja/blob/5a4bfd00863c03dae232d65746a93a7401020368/os/qml/workers/utils.js#L6

With the demo, I'm getting an object with a data-key containing a JSON string, like:
`{ data: '{"name":"ipv6"}'}`

Should we parse the string here or is it a job if the NodeWorker?

lahdekorpi, 2017-11-20:
Just to confirm, adding:

```js
if(typeof e === "object" && e.data === "string") {
  try {
    e.data = JSON.parse(e.data);
  } catch(e) {
    console.error("Could not parse messageData", e);
  }
}

Works perfectly and the demo starts working (shows the IP address).

WebReflection, 2017-11-20:
version 0.2.0 of node-worker does two things:

  * it solves this issue parsing data on the server too
  * it uses [circular-json](https://www.npmjs.com/package/circular-json) on both client and server to send/receive data

hope this works better for you now.

closed 2017-11-20 by WebReflection