WebReflection/benja #47: Should npm start be run as the alarm user instead of root?

lahdekorpi, 2017-11-21:
https://github.com/WebReflection/benja/blob/cf9a159ceb5ed3c8cf4f5ae10a901fad5d2eff87/os/qml/setup-user#L29

I've modified mine to skip npm (for extra speed) as well as sudo, and run everything as the alarm user. And everything works fine.  
Was there a reason for running via sudo? I've moved the app from vfat to ext4, did the fs and its permissions have something to do with this? Or I2C access?

Not trying to be whiny or anything, just curious...

WebReflection, 2017-11-21:
> run everything as the alarm user. And everything works fine.

that's because you're probably not using johnny-five or other IoT related libraries

lahdekorpi, 2017-11-21:
Yeah, I just realized this after reading up some more...

The qml command could still be spawned up as the alarm user. Although in the use cases this is intended for, it probably won't matter if it's run as root...

closed 2017-11-21 by lahdekorpi

WebReflection, 2017-11-21:
Thing is ... sudo should never be your bottleneck to anything 😅