WebReflection/benja #49: Raspberry PI2 fails to boot

barisbalic, 2017-12-26:
Hi there, really love the look of this project, exactly what I was hoping for.  On the downside I haven't been able to get it to boot all day.

I've been using `dd` on OSX to write the cards (like any other distro), and can get my RPI2 to boot Raspbian with any of the three SD cards I have.  However as soon as I try booting BENJA I just get stuck on a rainbow boot screen, I believe because the kernel isn't loading/can't be found.

I grabbed the latest image from the homepage.  Wondering if something is wrong, as I have tried...
- Multiple RPI2s
- Multiple SD cards
- Different blocksizes when writing the card

And I can't think of anything else :(

rip3rs, 2018-02-18:
I am on the same issue,

RPI 3 model B nothing happens, when trying to flash the sd card via dd.

Was you able to get it to work?

best

rip3rs, 2018-02-18:
This was clearly a "knowledge on transferring images to sf card.

I was unounting and focusing dd on disk2s1, which that is not the correct way.

I followed then the official way: https://www.raspberrypi.org/documentation/installation/installing-images/mac.md

these were my steps through terminal:
`⇒  diskutil list`
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *7.9 GB     disk2
   1:                 DOS_FAT_32 BENJA                   7.9 GB     disk2s1

`⇒  diskutil unmountDisk /dev/disk2`
`⇒  sudo dd bs=1m if=benja-rpi2.iso of=/dev/rdisk2 conv=sync`

and done :D

WebReflection, 2018-02-22:
I really should put a **just use [etcher](https://etcher.io)** info somewhere

closed 2018-02-22 by WebReflection