WebReflection/benja #51: Wifi Connection

AndreasWebdev, 2018-01-18:
Hello,

I have a raspi3b running BENJA successfully. How would I establish a wifi connection though? From my understanding, the documentation expects to run via lan (because there was no wifi setup).

WebReflection, 2018-01-18:
You have the [node option](https://www.npmjs.com/package/node-wifi), or the OS one:

```sh
# verify wifi path
iw dev # or ip addr

# wifi path will look similar to wlp2s0
# use such name after wifi-menu
wifi-menu wlp2s0

# choose your WiFi network and enter password

# verify you are connected
ping 8.8.8.8

# PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
# 64 bytes from 8.8.8.8: icmp_seq=1 ttl=53 time=8.15 ms
# 64 bytes from 8.8.8.8: icmp_seq=2 ttl=53 time=9.36 ms
# 64 bytes from 8.8.8.8: icmp_seq=3 ttl=53 time=7.92 ms
# ...
# press Ctrl+C to stop the ping

# if not, try to start the service manually
#> netctl start wlp2s0-YourWiFi-NAME
```

To start it automatically on boot: `netctl enable wlp2s0-YourWiFi-NAME`

closed 2018-01-18 by WebReflection

AndreasWebdev, 2018-01-20:
Hey, my wlan Interface is called wlan0, scanning with wifi-menu seems to fail with "No networks found" even though there are 5 networks around me.

WebReflection, 2018-01-21:
via ethernet, tru to update the system first via `sudo pacman -Syu` then try again