WebReflection/benja #53: Sound output Raspberry Pi 3

Jonas-SfirLabs, 2018-01-29:
First of all, thank you Andrea for your amazing work.

I am having troubles changing the audio output in Benja. I want to force audio output on the headphone jack rather than through HDMI. In Raspian I did that by changing it with alsamixer and the command:
`sudo amixer cset numid=3 1`

But this method does not work in Benja. I installed alsamixer and are also able to change volume with it but I cannot seem to change the output even though I do not get an error message when executing the code above.
Is there any other way to change the output?

I would appreciate any help. Thanks.

WebReflection, 2018-01-29:
have you tried via /boot/config.txt ?

https://www.raspberrypi.org/forums/viewtopic.php?t=23407

Jonas-SfirLabs, 2018-01-30:
Thanks for the quick answer. I tried multiple things in the config but it seems that there is only one way working in Benja.

`hdmi_drive=1`
together with
`hdmi_ignore_edid_audio=1`
did the trick for me. 
Thanks again for the tip.

closed 2018-01-30 by Jonas-SfirLabs